rp-rs/rp-hal-boards #9: How to run pico-compiled elf in `qemu-system-arm`

xphoniex, 2023-02-14:
Using [pico_usb_serial](https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/examples/pico_usb_serial.rs) example, I'm running:

```bash
$ qemu-system-arm --version
QEMU emulator version 6.2.0 (Debian 1:6.2+dfsg-2ubuntu6.6)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers

$ qemu-system-arm -M microbit ../../target/thumbv6m-none-eabi/release/examples/pico_usb_serial.elf -nographic -serial tcp::5678,server=on
WARNING: Image format was not specified for '../../target/thumbv6m-none-eabi/release/examples/pico_usb_serial.elf' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
QEMU 6.2.0 monitor - type 'help' for more information
(qemu) qemu-system-arm: -serial tcp::5678,server=on: info: QEMU waiting for connection on: disconnected:tcp:0.0.0.0:5678,server=on
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=00000000 R01=00000000 R02=00000000 R03=00000000
R04=00000000 R05=00000000 R06=00000000 R07=00000000
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=ffffffe0 R14=fffffff9 R15=00000000
XPSR=40000003 -Z-- A handler
FPSCR: 00000000
fish: Job 1, 'qemu-system-arm -M microbit ../…' terminated by signal SIGABRT (Abort)
```

and connecting to it in another terminal:

```bash
$ telnet localhost 5678
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Connection closed by foreign host.
```

What's the correct way?

ithinuel, 2023-02-19:
As far as I know there is no support for the rp-pico in [qemu](https://gitlab.com/qemu-project/qemu/-/tree/master/hw/arm).

xphoniex, 2023-02-19:
is there an alternative to qemu for simulating pico?

9names, 2023-02-19:
EDIT3: you're passing an elf file, but it expects a raw binary.

> is there an alternative to qemu for simulating pico?

https://github.com/wokwi/rp2040js

xphoniex, 2023-02-19:
> EDIT3: you're passing an elf file, but it expects a raw binary.
> 

how do I get the raw binary? it's what `rustc` is giving me, I only renamed it to `.elf`

9names, 2023-02-19:
rustc is giving you an elf. raw is not the same as uf2.

it's actually more than I think i can explain in a github comment, if you google: "what as an elf file" and "what is a uf2 file" you'll get pretty reasonable explanations.

to get a raw binary, you use the tool `objcopy` (or your target-specific version) on the elf file, with the format you want to get out.

```system
objcopy your_project.elf -O binary your_project.bin
```

xphoniex, 2023-02-19:
> 
> to get a raw binary, you use the tool `objcopy` (or your target-specific version) on the elf file, with the format you want to get out.
> 
> ```
> objcopy your_project.elf -O binary your_project.bin
> ```

```bash
$ objcopy target/thumbv6m-none-eabi/release/examples/pico_usb_serial.elf -O binary pico_usb_serial.bin
objcopy: Unable to recognise the format of the input file `target/thumbv6m-none-eabi/release/examples/pico_usb_serial.elf'
```

also
```bash
objcopy: supported targets: elf64-x86-64 elf32-i386 elf32-iamcu elf32-x86-64 pei-i386 pe-x86-64 pei-x86-64 elf64-l1om elf64-k1om elf64-little elf64-big elf32-little elf32-big pe-bigobj-x86-64 pe-i386 srec symbolsrec verilog tekhex binary ihex plugin
```

I don't see arm in there, do I need to a different version of `objcopy`?

9names, 2023-02-19:
correct. you can either use the arm-none-eabi-objcopy (if you have it), or you can install cargo-binutils so you can get cargo to run it for you:

here's the cargo-binutils approach:
```system
cargo install cargo-binutils
rustup component add llvm-tools-preview
cargo objcopy --release -- -O binary your_project.bin
```

xphoniex, 2023-02-19:
installed `arm-none-eabi-objcopy`, made the binary file, copied it over to rp2040js's root, renamed the bin file to `hello_uart.hex` which is needed by the script ran by `npm start`:

```
$ npm start
...
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024cca4 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024cca6 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024cca8 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccaa is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccac is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccae is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccb0 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccb2 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccb4 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[CortexM0Core] Warning: Instruction at 1024ccb6 is not implemented yet!
```

9names, 2023-02-19:
no, bin is needed for qemu.
if you want to make an intel hex file for rp2040js, you tell objcopy to do that for you:

```system
cargo objcopy --release -- -O ihex hello_uart.hex
```

xphoniex, 2023-02-19:
I already tried the bin one with qemu, failed with same error. 

Tried with `ihex` for rp2040js, almost same error but a bit different:
```
[CortexM0Core] Opcode: 0xffff (0xffff)
[RP2040] Read from invalid memory address: 2008954c
[RP2040] Read from invalid memory address: 2008954c
[CortexM0Core] Warning: Instruction at 2008954c is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[RP2040] Read from invalid memory address: 2008954c
[RP2040] Read from invalid memory address: 20089550
[CortexM0Core] Warning: Instruction at 2008954e is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[RP2040] Read from invalid memory address: 20089550
[RP2040] Read from invalid memory address: 20089550
[CortexM0Core] Warning: Instruction at 20089550 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[RP2040] Read from invalid memory address: 20089550
[RP2040] Read from invalid memory address: 20089554
[CortexM0Core] Warning: Instruction at 20089552 is not implemented yet!
[CortexM0Core] Opcode: 0xffff (0xffff)
[RP2040] Read from invalid memory address: 20089554
[RP2040] Read from invalid memory address: 20089554
```

9names, 2023-02-19:
It still looks like you're feeding it something invalid.
should look more like this:
```system
npm start

> rp2040js@0.17.8 start
> ts-node demo/emulator-run.ts

RP2040 GDB Server ready! Listening on port 3333
18:34:55.887 [SSI] Unimplemented peripheral write to 8: 0
18:34:55.887 [SSI] Unimplemented peripheral write to 14: 4
18:34:55.887 [SSI] Unimplemented peripheral write to f0: 1
18:34:55.887 [SSI] Unimplemented peripheral write to 0: 458752
18:34:55.887 [SSI] Unimplemented peripheral write to 8: 1
18:34:55.888 [SSI] Unimplemented peripheral write to 8: 0
18:34:55.888 [SSI] Unimplemented peripheral write to 0: 6226688
18:34:55.888 [SSI] Unimplemented peripheral write to 4: 0
18:34:55.888 [SSI] Unimplemented peripheral write to f4: 8737
18:34:55.888 [SSI] Unimplemented peripheral write to 8: 1
18:34:55.888 [SSI] Unimplemented peripheral write to 8: 0
18:34:55.888 [SSI] Unimplemented peripheral write to f4: 2684362786
18:34:55.888 [SSI] Unimplemented peripheral write to 8: 1
18:34:55.889 [XOSC_BASE] Unimplemented peripheral write to 0: 2720
18:34:55.889 [XOSC_BASE] Unimplemented peripheral write to c: 46
18:34:55.889 [XOSC_BASE] Unimplemented peripheral write to 0: 16429056
18:34:55.889 [XOSC_BASE] Unimplemented peripheral read from 4
18:34:55.889 [WATCHDOG_BASE] Unimplemented peripheral write to 2c: 524
18:34:55.889 [CLOCKS_BASE] Unimplemented peripheral write to 78: 0
18:34:55.889 [PLL_SYS_BASE] Unimplemented peripheral write to 4: 45
18:34:55.889 [PLL_SYS_BASE] Unimplemented peripheral write to 8: 0
18:34:55.889 [PLL_SYS_BASE] Unimplemented peripheral write to 0: 1
18:34:55.889 [PLL_SYS_BASE] Unimplemented peripheral write to 8: 125
18:34:55.889 [PLL_SYS_BASE] Unimplemented peripheral read from 4
18:34:55.890 [PLL_SYS_BASE] Unimplemented peripheral write to 4: 4294967262
18:34:55.890 [PLL_SYS_BASE] Unimplemented peripheral read from 0
18:34:55.890 [PLL_SYS_BASE] Unimplemented peripheral write to c: 401408
18:34:55.890 [PLL_SYS_BASE] Unimplemented peripheral read from 4
18:34:55.890 [PLL_SYS_BASE] Unimplemented peripheral write to 4: 4294967287
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 4: 45
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 8: 0
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 0: 1
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 8: 40
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral read from 4
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 4: 4294967262
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral read from 0
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to c: 335872
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral read from 4
18:34:55.890 [PLL_USB_BASE] Unimplemented peripheral write to 4: 4294967287
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 34
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 34
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 34: 256
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 40
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 40
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 40: 256
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 58
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 54
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 54: 4294965247
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 54
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 54: 4294967071
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 54
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 54: 4294967295
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 58
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 58: 256
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 64
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 60
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 60: 4294965247
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 60
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 60: 4294967071
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 60
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 60: 4294967295
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 64
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 64: 256
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 70
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 6c
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 6c: 4294965247
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 6c
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 6c: 4294967071
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral read from 6c
18:34:55.890 [CLOCKS_BASE] Unimplemented peripheral write to 6c: 4294967295
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral read from 70
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral write to 70: 262144
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral read from 48
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral write to 48: 4294965247
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral read from 48
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral write to 48: 4294967071
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral read from 48
18:34:55.891 [CLOCKS_BASE] Unimplemented peripheral write to 48: 4294967295
18:34:55.891 [UART0] Unimplemented peripheral write to 24: 67
18:34:55.891 [UART0] Unimplemented peripheral write to 28: 52
18:34:55.891 [UART0] Unimplemented peripheral write to 48: 3
UART example
value: 00
value: 01
value: 02
value: 03
value: 04
value: 05
value: 06
value: 07
value: 08
value: 09
value: 10
```

xphoniex, 2023-02-19:
can you get [pico_usb_serial](https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/examples/pico_usb_serial.rs) example running? doesn't work for me.

sadly, might have to switch to [esp32](https://esp-rs.github.io/book/tooling/simulating/qemu.html) against my wish, pico doesn't even have as much support as avr. :confused:

9names, 2023-02-19:
not sure, i did not test USB CDC as that's much more complex. i tested with uart.

if you do need a functional emulator esp32 is a good choice! espressif put lots of effort into getting it working well.

ithinuel, 2023-02-22:
> is there an alternative to qemu for simulating pico?

Considering how inexpansive the rp2040 based boards usually are (like rp-pico), the best alternative to a simulator IMHO is to get two of them. Use one as the debug probe and work with the physical board :)

Also, thanks to the [embedded-hal](https://crates.io/crates/embedded-hal), you can develop your application regardless of the actual hardware and switch later.

9names, 2024-08-23:
Closing this issue as out-of-scope. It looks like the OP worked out how to run cortex-m code in qemu in https://github.com/rust-embedded/cortex-m-quickstart/issues/119 anyway

closed 2024-08-23 by 9names