rp-rs/rp-hal-boards #38: rp2040/pico W pins to talk to CYW43?

brandonros, 2023-08-22:
From the datasheet:

[image]

Which I think would translate to this in code:

```rust
 let (mut pio, sm0, _, _, _) = pac.PIO0.split(&mut pac.RESETS);
    let mut spi: spi_pio::Spi<'_, _, _, _, _, _, 8> = spi_pio::Spi::new(
        (&mut pio, sm0),
        (pins.gpio24, pins.gpio24, pins.gpio29), // (miso, mosi, sclk)
        embedded_hal::spi::MODE_0,
        16u32.MHz(),
        clocks.peripheral_clock.freq(),
    )
    .ok()
    .unwrap();
    let spi_cs = pins.gpio25.into_push_pull_output();
```

but these pins are "not found"?

brandonros, 2023-08-22:
I see now the names are:

```rust
let (mut pio, sm0, _, _, _) = pac.PIO0.split(&mut pac.RESETS);
    let mut spi: spi_pio::Spi<'_, _, _, _, _, _, 8> = spi_pio::Spi::new(
        (&mut pio, sm0),
        (pins.vbus_detect, pins.vbus_detect, pins.voltage_monitor), // (miso, mosi, sclk)
        embedded_hal::spi::MODE_0,
        16u32.MHz(),
        clocks.peripheral_clock.freq(),
    )
    .ok()
    .unwrap();
    let spi_cs = pins.led.into_push_pull_output();
```

But it seems like MOSI + MISO being the same pin isn't supported?

```
 $ cargo build
   Compiling my-cyw43-driver v0.1.0 (/Users/brandon/Desktop/my-cyw43-driver)
warning: unused variable: `spi_cs`
  --> src/main.rs:59:9
   |
59 |     let spi_cs = pins.led.into_push_pull_output();
   |         ^^^^^^ help: if this is intentional, prefix it with an underscore: `_spi_cs`
   |
   = note: `#[warn(unused_variables)]` on by default

error[E0382]: use of moved value: `pins.vbus_detect`
  --> src/main.rs:52:28
   |
52 |         (pins.vbus_detect, pins.vbus_detect, pins.voltage_monitor), // (miso, mosi, sclk)
   |          ----------------  ^^^^^^^^^^^^^^^^ value used here after move
   |          |
   |          value moved here
   |
   = note: move occurs because `pins.vbus_detect` has type `rp_pico::rp2040_hal::gpio::Pin<Gpio24, rp_pico::rp2040_hal::gpio::Disabled<rp_pico::rp2040_hal::gpio::PullDown>>`, which does not implement the `Copy` trait
```

Will keep seeing if I can workaround it.

jannic, 2023-08-22:
AFAIK the hardware SPI implementation of the rp2040 doesn't support this mode.
Two possible solutions:
a) easy but slow, bit-bang SPI. https://github.com/rp-rs/rp-hal/pull/380/files#diff-2f2f98170fea596b486eaa0345625a8eea400be05746cf90ffa1df2577114b3b
b) use PIO: https://github.com/embassy-rs/embassy/blob/main/cyw43-pio/src/lib.rs

closed 2023-08-22 by brandonros