rp-rs/rp-hal-boards #54: trying to build blinky for pi pico

HeadhunterXamd, 2024-02-02:
I get a strange error when trying to build the blinky example:

```rust
error[E0433]: failed to resolve: use of undeclared crate or module `embedded_hal`
  --> src/main.rs:17:5
   |
17 | use embedded_hal::digital::v2::OutputPin;
   |     ^^^^^^^^^^^^ use of undeclared crate or module `embedded_hal`

```

and a few lines further:
```rust
error[E0599]: no method named `set_high` found for struct `rp_pico::rp2040_hal::gpio::Pin` in the current scope
  --> src/main.rs:85:17
   |
85 |         led_pin.set_high().unwrap();
   |                 ^^^^^^^^ method not found in `Pin<Gpio25, FunctionSio<SioOutput>, PullDown>`
   |
  ::: /xx/xx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-0.2.7/src/digital/v2.rs:60:8
   |
60 |     fn set_high(&mut self) -> Result<(), Self::Error>;
   |        -------- the method is available for `rp_pico::rp2040_hal::gpio::Pin<Gpio25, FunctionSio<SioOutput>, PullDown>` here
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
14 + use embedded_hal::digital::v2::OutputPin;
   |

```

I tried to use the toml of the example and the toml based on the [from scratch](https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/README.md#from-scratch) chapter.

I tried this with the example code from the blinky file.

jannic, 2024-02-02:
For blinky, you need to add a dependency on `embedded_hal = "0.2"`.
Sorry, the docs could explain that better.

TheGhostHuCodes, 2024-02-04:
I ran into this issue a few days ago as well. Is there any plans to update to `embedded_hal` 1.0? Is this a good task for someone new to this project to try doing?

jannic, 2024-02-04:
In this case, the issue was not related to the specific version of embedded_hal. The mentioned docs just don't explain that you probably need to add a dependency on embedded_hal when you extend the trivial example.

Regarding version 1.0: rp2040-hal is mostly ready on the main branch, and there are open merge requests for the missing parts. Updating the board crates wasn't started yet (as far as I am aware).

TheGhostHuCodes, 2024-02-06:
Would it be reasonable for `rp_pico` crate to re-export its `embedded_hal` so that the user of `rp_pico` wouldn't need to know which version to install?

ithinuel, 2024-02-06:
Not really imho. Both would be equivalently supported.
The "limiting" factor for a user would likely be the external peripheral drivers they're using instead.

jannic, 2024-02-06:
Maybe this is another data point that it would be reasonable to provide inherent implementations of all fundamental operations, especially easy things like setting a gpio pin? That way, in a simple firmware, it would no longer be required to import the embedded-hal traits.

In the past I've held the opinion that any serious firmware is structured in a way that separates setup code working on the MCU specific inherent functions from driver code working on the embedded-hal traits. Even if the driver code is not in a separate crate but specific to this single firmware. But perhaps that's an overly complex, and not sufficiently beginner-friendly approach?