rp-rs/rp-hal-boards #57: Upgrade to embedded-hal v1.0.0

Michael2MacDonald, 2024-02-23:
Upgrade to `embedded-hal` `v1.0.0`

`v1.0.0` is the most current release of `embedded-hal` and has several breaking changes compared to `v0.2.x`. One example is `v1.0.0` replaces `embedded_hal::digital::v2::OutputPin` with `embedded_hal::digital::OutputPin`. `v1.0.0` also drops the adc interface and the timer CountDown trait. Moving forward, most code will be written for `v1.0.0` and newer versions. The whole point of `embedded-hal` is to provide interfaces for portable and hardware-agnostic code. Using an outdated version defeats its entire purpose.

`rp-hal` and `rp2040-hal` already use `embedded-hal` `v1.0.0` to my knowledge.

jenshnielsen, 2024-03-31:
For the record this was merged to main in https://github.com/rp-rs/rp-hal-boards/pull/60

9names, 2024-04-21:
Resolved by #60, and as of #62 the published versions of the crates on crates.io now support 1.0

closed 2024-04-21 by 9names