rp-rs/rp-hal-boards #58: Issue(s) with / ideas for the from-scratch section

exo-cortex, 2024-02-27:
An issue I found is that when I add funtionality like in "pico_blinky.rs" I get errors, because "cortex_m" is not found and and "embedded_hal::digital::v2::OutputPin" does not work, because there seems to be no "v2" in there. (was there an update maybe that made parts of the code obsolete here?)

Another idea/suggestion i have for the *from scratch* example: it would be nice to include a directory tree of the from-scratch example (so there is also the memory.x and .cargo/config.toml) so it's easier to understand everything that is involved.

It would be nice for beginners if there would be one very simple example that works so that one can start from there. 

I am unsure about which Cargo.toml is the one for pico_blinky.rs. I know there is the rp2040 project template, but that already seems to be very complicated compared to the from scratch-example. Especially the Cargo.toml's are really dense (what does "... = { workspace = true, optional = true } mean in this context?) 
Would it be possible to have just the most basic program explained completely somewhere? 
I feel the examples share the same Cargo.toml and therefor a lot of dependencies are not needed for

exo-cortex, 2024-02-27:
Hm that would work as well. I can try that.
I started "from scratch" as that was a section in the Readme. 
Can you recommend a template - there is one, but as there are so many different repositories and interconnected Hal-related crates I am sometimes unsure about which are needed from which. Maybe the readme could also offer a link to the recommended base-template for the raspberry pi pico. 

but thank you very much!

closed 2024-02-27 by exo-cortex

9names, 2024-02-27:
https://github.com/rp-rs/rp2040-project-template

ithinuel, 2024-02-28:
> Remember to build and run every time you remove something so you know what you need

If you’re not already using it, I recommend [cargo watch](https://crates.io/crates/cargo-watch) for that purpose.
Fire `cargo watch -x build` from your project in a terminal, and enjoy modifying your files and see it being rebuilt any time you save.