rp-rs/rp-hal-boards #59: Add support for RP2040 with 16MB of flash

Fede-26, 2024-03-11:
I want to add support for this board: [https://it.aliexpress.com/item/1005003657583800.html](https://it.aliexpress.com/item/1005003657583800.html).
It's a clone of the official pi pico with 16MB of flash.

I can't find how to specify the ROM size like the (pimoroni-pico-lipo-16mb)[https://github.com/rp-rs/rp-hal-boards/tree/main/boards/pimoroni-pico-lipo-16mb].

I believe this is the only change necessary, because when using the C sdk, I just need to copy the official board config and change the ROM size.

jannic, 2024-03-18:
I don't have a board with 16MB flash to test this, but as far as I know, all you need to do is update `memory.x` accordingly.
However, there's currently no mechanism to have a BSP specific memory.x.

In fact, it should still work if you don't update it. You just wouldn't be able to use all that flash, but only the first 2MB.

Fede-26, 2024-03-19:
Then why the readme (and the specific documentation in the relative directory) says

> Note that if you use this crate the compiler will expect the full 16MB flash space, and so it may not work if you only have the 4MB variant.

?

If you can't change the flash size of each different bsp shouldn't the compiler expect a normal 2MB flash?

ChocolateLoverRaj, 2024-11-28:
I have a black RP2040 16MB board from AliExpress. I would like to add the board to this repo, but I'm not sure how. I don't see any documentation about the board. It seems to be like the official Raspberry Pi Pico but it has a RGB LED connected to pin 23. Both the `rp-pico` `pico_ws2812_led` and `waveshare_rp2040_zero_neopixel_rainbow` examples work if I set the RGB pin to 23.