rp-rs/rp-hal-boards #76: Using the Badger2040 example with the Badger 2040W?

shymega, 2024-07-24:
Hi,

I'm looking to write firmware for a Badger 2040W, based on [this](https://github.com/rp-rs/rp-hal-boards/tree/main/boards/pimoroni_badger2040) example, but using the Embassy framework.

I was just curious if the example needs adjusting (GPIO, RTC, or other variables) for the 2040W.

Thanks.

ithinuel, 2024-07-24:
Hello, thank you for reaching out.

@Dirbaio I think you might be able to answer this :)

9names, 2024-07-26:
Assuming badger2040w has the same pinout as badger2040, this example would be a good starting point.
https://github.com/9names/uc8151-rs/tree/main/examples/async

shymega, 2024-09-04:
So, I've found [this example](https://github.com/rp-rs/rp-hal-boards/tree/main/boards/pimoroni_badger2040), and I think it's adaptable for the Badger RP2040W.

I've flashed it, and it works with the e-ink display. I could make a PR for an example with the RP2040W scanning WiFi SSID broadcasts.

Just a quick Q - do we have support for wireless connectivity with `rp-hal` on the RP2040W?

jannic, 2024-09-04:
No, we don't. Wifi/Networking is much easier to support with async support, and the async story of rp-hal is not yet complete. So for the Pico-W we usually suggest embassy.

shymega, 2024-09-04:
I was actually planning to use Embassy (for my Badger, not the example), so that's good to know. For the example, would you prefer I keep it simple, and ignore the wireless functionality?