rp-rs/rp-hal-boards #78: pimoroni-tufty2040 - either crate trying to compile with std

nebelgrau77, 2024-08-04:
I'm trying to build something using the pimoroni tufty2040 crate. 

It works fine with `cargo build --release --example <name>`, but when I try to replicate the exact same example from scratch (correct config.toml, correct memory.x, necessary dependencies) it won't compile with the following error:

_error[E0463]: can't find crate for `std`
  --> /home/nebelgrau/.cargo/registry/src/index.crates.io-6f17d22bba15001f/either-1.13.0/src/lib.rs:19:1
   |
19 | extern crate std;
   | ^^^^^^^^^^^^^^^^^ can't find crate
   |
   = note: the `thumbv6m-none-eabi` target may not support the standard library_
   
Somehow it's trying to compile the either crate with std support, but I have no clue where this originates. 

Here's my Cargo.toml: 
 
```
[dependencies]
pimoroni-tufty2040 = "0.1.0"
rp2040-hal = "0.3.0"
rp2040-boot2 = "0.2.0"
embedded-hal = { version = "0.2.5", features = ["unproven"] }
panic-halt = "0.2.0"
cortex-m-rt = "0.7"
cortex-m = "0.7.2"
embedded-graphics = "0.7.1"
st7789 = "0.6.1"
```

What am I missing here, does a nostd feature need to be added somewhere?

jannic, 2024-08-04:
> What am I missing here, does a nostd feature need to be added somewhere?

I can't be sure from your problem description alone, but I guess your `Cargo.toml` is missing the `edition = "2021"` entry.

Without that, cargo defaults to an older dependency resolver implementation, which unifies crate features between `dependencies` and `build-dependencies`. As the build-dependencies are meant to run on the host operating system, ie. with `std` available, the `use_std` feature of the `either` crate gets activated. That crate is also used inside the firmware, but there `std` is not available, leading to the error message you see.

With `edition = "2021"` resolver version 2 gets activated. See https://doc.rust-lang.org/cargo/reference/resolver.html#feature-resolver-version-2 for details. The important difference is: "Features enabled on [build-dependencies](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#build-dependencies) or proc-macros will not be unified when those same dependencies are used as a normal dependency."

nebelgrau77, 2024-08-04:
That was it, thank you! I didn't realize how much the edition mattered in this case. There was also an error in my Cargo.toml, as embedded-hal is of course supposed to be 1.0.0. Now it works, thank you!

closed 2024-08-04 by nebelgrau77