rp-rs/rp-hal-boards #80: Arduino Nano RP2040 Connect blinky example not working

alanbowman, 2024-08-09:
I'm trying to get set up with the Arduino Nano RP2040 Connect board, but can't get the blinky example to work.  I've tested that the Arduino IDE can program its own blinky example (so the board/LED definitely works) and I've tried the pico_blinky example on a Pico board and seen the flashing LED there (so I think my toolchain's OK).  However, when I run `cargo run --release --example nano_blinky` I can see that the UF2 is copied to my board, which then reboots (so I lose the drive that it appears as in bootloader mode) but there's no sign of a flashing LED.

I don't have the means to connect a debugger to the board, and I don't see any indication of any signs of life.  Can anyone suggest where I might be going wrong?

9names, 2024-08-09:
The example says it blinks an LED attached to GPIO25, but there's no LED attached to that pin.
As far as I can tell via the schematic, the LEDs are all controlled by the wifi-nina module: https://docs.arduino.cc/resources/schematics/ABX00053-schematics.pdf
If you have an oscilloscope or multimeter you can probably see the GPIO25 pin changing state, or you could hook up a current limiting resistor and LED to another free pin to verify that code is running.

alanbowman, 2024-08-09:
The comment refers to GPIO25 but it looks like it actually uses SPICLK/SCK0, which does have an LED attached.  I *think* the comment is incorrect but I'll dig out a multimeter and check anyway.

alanbowman, 2024-08-09:
Progress!  In the lib.rs file for the Arduino board, there's some lines that select the bootloader.  The generic one is enabled and a specific one is commented out.  Swapping these over and recompiling gives me the blinky LED I'd expect:

This works (for me):
```
#[cfg(feature = "boot2")]
#[link_section = ".boot2"]
#[no_mangle]
#[used]
// pub static BOOT2_FIRMWARE: [u8; 256] = rp2040_boot2::BOOT_LOADER_GENERIC_03H;
pub static BOOT2_FIRMWARE: [u8; 256] = rp2040_boot2::BOOT_LOADER_AT25SF128A;
```

I recall some comments that different flash chips were used during the chip shortages, but I'm not sure which I have or why the generic option didn't work.  I'm happy to test out any suggestions people have to make this work without modifications.

9names, 2024-08-10:
BOOT_LOADER_GENERIC_03H should be the most forgiving boot2 implementation, it's surprising that it doesn't work here.
The schematic says the SPI flash is AT25SF128A-MHB-T, so if that boot2 works for you we should just change to that in the BSP. A PR doing this would be welcome.

alanbowman, 2024-08-12:
I've checked and my board does have the AT25SF128A flash chip so that makes sense.  However, having looked through the history of the various related repos I see that there's already been a move from specific->generic because of different flash chips being used (https://github.com/rp-rs/rp-hal/issues/503).  Given that, I won't raise a PR to change it back.  I'll try any suggestions to understand the problem with the generic bootloader, though.

raspiantoro, 2024-08-28:
I'm able to run `cargo run --release --example nano_blinky`, and the LED works as expected. However, the usb port on my arduino nano connect no longer works. Is this expected behavior or a bug?