rp-rs/rp-hal-boards #89: Second-Stage Bootloader for Feather RP2040

Tosainu, 2024-09-29:
Hello.

I wonder why `adafruit-feather-rp2040` uses `rp2040_boot2::BOOT_LOADER_GD25Q64CS` as the second-stage boot loader. According to the [schematic](https://learn.adafruit.com/adafruit-feather-rp2040-pico/downloads#schematic-and-fab-print-3132541) of Adafruit Feather RP2040, the flash chip on this board is W25Q64JV which has a similar instruction set as W25Q80DV. Is there any reason for using it over `rp2040_boot2::BOOT_LOADER_W25Q080`?
https://github.com/rp-rs/rp-hal-boards/blob/7431e73c51a89b2fe81afb51896b2530afd40b8c/boards/adafruit-feather-rp2040/src/lib.rs#L10-L16

As a side note:
- Their factory-test program uses `boot2_w25qs080.S`:  
https://github.com/adafruit/Adafruit-Feather-RP2040-PCB/blob/ea88166891ee0a1697a3899a5d55ab3722a2f125/feather_test/CMakeLists.txt#L32
- Interestingly, both `BOOT_LOADER_W25Q080` and `BOOT_LOADER_GD25Q64CS` work on my Feather RP2040 board without any notable issues (at least they reach my code eventually).

9names, 2024-09-29:
The schematic you reference is Rev D of this board.
All previous versions of the schematic for this board do not list which flash chip was used. During the global chip shortage boards often changed components based on what they could get.

Here's where that BOOT2 variant was added (specifically for the adafruit feather rp2040, because that was what was on the board)
https://github.com/rp-rs/rp2040-boot2/pull/9
And where initial support for this board was added:
https://github.com/rp-rs/rp-hal/pull/91

Tosainu, 2024-09-29:
Thanks for the kind explanation! I double-checked my board and yes, the flash is GD25Q64CS.

[image]

closed 2024-09-29 by Tosainu