rp-rs/rp-hal-boards #100: GPIO pins on seeeduino XIAO do the opposite of what they're told to do.

noahmartinwilliams, 2024-12-31:
When I try to set the pin to high it goes low and vice versa. I've tested this out on the LED pins, and a GPIO pin. I'm new to rust, and I'm not sure if this is the repository to be filing the bug report at since there's so many packages in the dependency tree here. I've attached the code that I'm using to test this.
[tester.tar.gz](https://github.com/user-attachments/files/18283752/tester.tar.gz)

9names, 2025-01-01:
Have you looked at the schematic for that board?
https://files.seeedstudio.com/wiki/XIAO-RP2040/res/Seeed-Studio-XIAO-RP2040-v1.3.pdf

You can see the LEDs on page 2 are permanently connected to 3.3 volts on one side, and the other side is connected to the rp2040 pins through a current limiting resistor.
That means that current will only flow when the pin is being driven low.
If you're seeing the LEDs when setting pins low, that means the pins are working as expected.

noahmartinwilliams, 2025-01-01:
Ok, but I also tested it on a GPIO pin. I'll test it on the other GPIO pins later. The numbering system seems a little bit confusing, but I'm pretty sure that the a2 pin in the code is the third from the top down on the left of this image.
[image]

9names, 2025-01-01:
Yes, that's the correct pin.
How did you test it?
I just used your program with a xiao-rp2040. I measured with a multimeter and the pin goes to 3.3V for 2 seconds, then back to 0V and stays there.

noahmartinwilliams, 2025-01-01:
I used my own multi-meter which showed the exact opposite. It ends on 3.3 volts and stays there. Maybe there's something wrong with my particular chip? I'll test on a different one tomorrow, I guess.

closed 2025-01-01 by noahmartinwilliams